fermi-ad/acsys-python #4: Allow arguments to be passed in `acsys.run_client()`

rneswold, 2021-03-24:
It would be nice if the `run_client` function would have an arguments field where the user could pass arguments to the target async method in addition to the newly create connection object.

rneswold, 2021-03-31:
@kjhazelwood : Are you looking for `**kwargs` style? So a `main` function would look like this:

```python
async main(con, **args):
    # do stuff
```

and you would invoke it as:

```python
acsys.run_client(main, arg1='first', arg2='second')
```

Or the `*args` style?

kjhazelwood, 2021-03-31:
I dont think I have a preference, which ever is most convenient to implement.

rneswold, 2021-03-31:
Done: d288cfa95602b48f81012105b64fea62d9b021dc

This allows keyword arguments to be passed (they seemed the most flexible.) The next release will contain this feature.

closed 2021-03-31 by rneswold