fermi-ad/acsys-python #11: DPM data logger closes connection after retrieving only portion of entries requested

kjhazelwood, 2021-03-26:
When retrieving data for multiple devices from one common logger node, the DPM returns data for a portion of the devices and then without any explanation throws a 1-34 ACNET DISCONNECTED error and fails. @beauremus and @charlieking65 modified DPM02 to add a 20mS delay between responses, after which I was able to retrieve all the desired data all be it slower.

device count: 20
device name format: I:L1{230-314}
logger node: Minj2
logger event: P,1000,TRUE
start date: 2021-01-01 00:00:00
end date: 2021-02-01 00:00:00

beauremus, 2021-03-26:
I was wrong about the delay, it's 10ms. We went through several iterations to determine what was most reliable without being too slow.

Clearly, this manual delay is not desirable. The long-term solution is to have a TCP connection between the DPM and the client. Currently, we use ACNET/UDP to communicate between DPM and the TCP proxy. We believe that this is where packets are being dropped. Introducing the delay is enough time for the queues to flush without dropping packets.

beauremus, 2021-05-27:
We have alleviated this issue by rearchitecting the infrastructure on-site. The final solution is to have a TCP connection all the way to DPM, see #20.

closed 2021-05-27 by beauremus