fermi-ad/acsys-python #12: DPM data logger fields feel like they should be specifed in .add_entry(), not .start()

kjhazelwood, 2021-03-26:
Instead of specifying the logger node, start date and end date in dpm.start(), it seems more intuitive to specify this information when adding entries. This would also ease grabbing logged data from multiple nodes at once instead of creating a new dpm context for each unique logger node or date span.

rneswold, 2021-03-26:
Our initial intent was to have each `DpmContext` loop have an associated "model" (i.e. data source.) So a loop would return all its data from a save file, or the accelerator, or a data logger. The nice feature about this is you can use the same set of DRF requests but change the data source in the `.start()` method.

The other approach is to require every DRF string to specify its source.

@beauremus, @charlieking65, and I had several discussions about `acsys` recently, including this very issue. I think (and they can confirm or deny it) we decided that, if a user specifies a model in the `.start()` method, it would get applied to every DRF request that doesn't already specify a model. That should get us the best of both worlds.

beauremus, 2021-03-26:
I copied over the Redmine wiki page on [Data sources](/fermi-controls/acsys-python/wiki/Data-sources) and added some more context.

You can use both strategies now. The documentation wasn't clear on this.

beauremus, 2021-05-27:
This feature already existed. Closing. Feel free to reopen if this is not correct.

closed 2021-05-27 by beauremus

kjhazelwood, 2021-05-27:
I agree, thanks.