fermi-ad/acsys-python #13: Is there a way to destroy a DPM request?

kjhazelwood, 2021-03-26:
To address #11, I'm specifying DPM02 for my DPM context. I had a datalogger job fail overnight, no when I try to restart I'm given a [1 -6] error essentially saying the connection timed out or the job is already queued. Is there a way to tell DPM to destroy all jobs in the queue for the machine I'm on? If not, I think this would be very usefull.

kjhazelwood, 2021-03-26:
Actually, the error appears to be what you get if you are not on the right network to connect to DPM. I had assumed fgz was sufficient to connect to the DPM. I connected through vpn.fnal.gov and now it is connecting as expected. Not sure, If I tricked the DPM into thinking Im a new request or fgz isnt sufficient to DPM.

rneswold, 2021-03-26:
You can stop all your front-end requests by calling `dpm.stop()`. `dpm.start()` will restart all your requests. The connection to DPM isn't restarted, however. To do that you have to exit the `async with DpmContext` scope so things get cleaned up.

But even though there are these workarounds, we don't want our users to have to know about them or use them. The `dpm` module is supposed to recover from transient errors so I believe you found a bug. Unfortunately it might take a while to reproduce the error, so you might have to use the workarounds. ☹️

beauremus, 2021-03-26:
The fgz vs VPN event is a coincidence.
@charlieking65 found DPM02 crashed last night in a way that it wasn't taking requests anymore. He must have restarted in between you being on fgz and VPN. To be clear fgz should be fine for DPM access.

The [1 -6] ACNET_REQTMO is saying that it can't find DPM02.

In the future, this should be invisible as you failover to another DPM. I created #14 to make this even more seamless. At the moment, when switching DPMs the request will start again meaning that the developer needs to know that the switch has happened and throw collected data away and start over.

The need to specify DPM02 is driven by the fact that we have multiple DPM versions deployed at the moment. DPM02 has the newest code, but because you had this issue overnight we'll have to look at things closer before deploying it to all the DPMs. Once the newest code is deployed, setting `dpm_node=None` will enable service discovery to find the best DPM for you and failover in the case of an error like this.

rneswold, 2021-03-26:
> Once the newest code is deployed, setting dpm_node=None will enable service discovery to find the best DPM for you and failover in the case of an error like this.

I'm going to work on the backward-compatible Kerberos support this morning. Hopefully we can get this out soon.

kjhazelwood, 2021-03-26:
By chance, is DPM02 dead again. I re-ran my script again this morning and it stalled at the same spot.

rneswold, 2021-03-26:
> By chance, is DPM02 dead again? I re-ran my script again this morning and it stalled at the same spot.

This sort of question should be asked in Slack. 😄