fermi-ad/acsys-python #16: Support Kerberos authentication with service discovery

beauremus, 2021-03-26:
Support Kerberos authentication with service discovery. For a smooth rollout, this requires two authentication strategies in the client until we can deprecate the old version.

The existing clients are latched to `DPM01` by default. We can deploy a new incremental update that has the principal hardcoded for DPM01 and supports the new message type that tells the client what principal to use.

rneswold, 2021-03-26:
So the plan is
* Do service discovery to pick a DPM
* When Kerberos is needed, start the exchange
* If `Authenticate_reply` is received, continue successfully. Otherwise, if `Status_reply` is received, switch over to DPM01 and hardcode the service name.

rneswold, 2021-03-31:
Done: b8390026c96b6ecc4c6dad897808e9e5c35c557a

When switching from a DPM that doesn't do the proper Kerberos exchange, the runtime emit a message saying a `GeneratorExit` exception wasn't getting handled. I cannot figure out why it's reporting this. If anyone wants to try to figure it out, I'd appreciate it. It doesn't affect the script from running.

rneswold, 2021-03-31:
Once all DPMs use the new Kerberos protocol, the code in this commit can be simplified.

rneswold, 2021-05-19:
The "old" protocol is no longer used.

commit a4b78ef2 (2021-05-19): fix: remove old Kerberos support.

All DPMs are using the new protocol so this commit removes the block
that tried to detect the older protocol and compensate for it.

Close #16.
closed 2021-05-19 by rneswold