fermi-ad/acsys-python #19: Add a module to support DRF operations

rneswold, 2021-04-01:
`acsys` needs a module which understands DRF. There have already been a couple issues (#8, #14) which could be solved or supported if we could parse DRF strings.

This module could:

- Parse DRF strings into the various components
- Add a class hierarchy which models DRF requests. These classes' `.__str__()` methods would return DRF strings.

beauremus, 2021-04-01:
This is a good use case for a service 😉 
We should have **a** reference implementation that other applications can make queries to.
Reimplementing this in Python could lead to inconsistencies.

rneswold, 2021-04-01:
If it becomes a service:
* We can use protocol compiler to model DRF.
* DRF parsing will be slightly slower than native parsing, but everyone gets the same support.

If we add a parser to `acsys.drf`:
* We need to create a set of DRF examples that all implementations need to successfully parse.
* We need a document that identifies all parsers so their maintainers can be notified when DRF changes (for instance, only Charlie's Java parser supports some DRFv3 features that no other parser supports.)