fermi-ad/acsys-python #21: acsys.sync issue after upgrading to 0.10.0

kjhazelwood, 2021-04-20:
After upgrading to 0.10.0 from 0.9.10, `import acsys.sync` fails with the following:

```
File "/home/kjh/.local/lib/python3.6/site-packages/acsys/sync/__init__.py", line 6, in <module>
    import acsys.sync.syncd_protocol as syncd_protocol
AttributeError: module 'acsys' has no attribute 'sync'
```

rneswold, 2021-04-20:
Looks like c4f7a3f7072e57333a89c3f15cec23d1b6f61778 added that import line. It should probably just be

```python
import syncd_protocol
```

rneswold, 2021-04-20:
This also illustrates why we need a test suite to catch regressions. 😄

beauremus, 2021-04-20:
Sync was using `import acsys.sync.syncd_protocol as syncd_protocol` and because this definition was in the `__init__.py` it conflicted with the file named `synd_protocol.py` in the `sync` subpackage.

closed 2021-04-20 by beauremus

kjhazelwood, 2021-04-20:
This fixed the problem, thanks.