fermi-ad/acsys-python #23: Support ramp card settings

beauremus, 2021-05-07:
Ramp cards don't follow standard conventions. We need to develop methods for setting `f(t)` and `d(t)` values for CAMAC 400 family cards.

rneswold, 2021-05-18:
Added support (35f33659e546f31efcddfc4d22bda05dd723562b) for a scaling service (@charlieking65 implemented the service in Java.) We can support Jeff's need using this service and an additional function.

There is a minor issue in that he gets a `StopAsyncIteration` exception when he uses the additional function. In other, similar cases we needed to add a `try-except` block which catches and ignores the error. It would be nice to do this in the library so Jeff (and future users) wouldn't have to in their code.

rneswold, 2021-05-18:
Maybe this line: https://github.com/fermi-controls/acsys-python/blob/35f33659e546f31efcddfc4d22bda05dd723562b/acsys/dpm/__init__.py#L815 is where the `StopAsyncIteration` is escaping.

rneswold, 2021-05-18:
Maybe this: https://github.com/fermi-controls/acsys-python/blob/35f33659e546f31efcddfc4d22bda05dd723562b/acsys/dpm/__init__.py#L317 should be
```python
    return None
```
so it closes out the generator.

rneswold, 2021-05-19:
Jeff verified it's working. The exception warning is being tracked by another issue.

closed 2021-05-19 by rneswold