fermi-ad/acsys-python #24: Case should be consistent

beauremus, 2021-05-12:
Idiomatic Python uses snake case (I just realized how amusing that is 😁 ), but we have some methods that are camel case and some that are snake case. I personally don't think it matters which we choose, but it should be consistent. I believe that when we introduce linting into our continuous integration, it will catch things like this.

rneswold, 2021-05-12:
Do any of your proposed changes affect the public API? We may have to do a phased release where the first one has extra functions (in the proper case) and the older ones marked deprecated (if Python has such a thing.) The next release we would remove the old functions.

Or we go to v1.0 which implies the API could change.

beauremus, 2021-05-12:
We should definitely consider implementing CI and lining for v1.

I believe this is what should be used to mark a method as deprecated. https://docs.python.org/3/library/exceptions.html#DeprecationWarning

rneswold, 2021-05-18:
I agree we should use snake-case for function and method names. What about property names? For instance, `acsys.status.Status` has computed properties like `.isFatal` and `.isSuccess`.

rneswold, 2021-05-18:
This commit includes changing the property names. There weren't too many areas to change, fortunately. I'm leaving this issue open for a while longer to get feedback from @beauremus and to do a little more testing. I did run it with my test scripts and didn't have any "unknown function" errors.

beauremus, 2021-05-27:
In commit 6a25e3b55462390e4c69d63346d7ab64be2d6478 I changed several variables to snake_case using `flake8` and `pylint` to validate my changes. The first goal is that our API be idiomatic Python. The second goal is to use tools like `flake8` and `pylint` to automate checking for snake_case across all our code. That means correcting errors raised by these tools.

beauremus, 2021-05-27:
Goal one is covered by this issue and complete. Goal two is covered by #29.

closed 2021-05-27 by beauremus