fermi-ad/acsys-python #25: More featureful data type checking

beauremus, 2021-05-12:
`.isReadingFor(0)` is a convenient method for checking that a response is for a particular `tag`.

What if I want to run the same method for multiple `tag`s?

I propose that `.isReadingFor` take n arguments that allow for as many `tag`s as the developer wants.

Also, if the same code is run no matter the tag `.isReadingFor` should accept zero arguments allowing for the type to be checked without consideration for the `tag`.

rneswold, 2021-05-12:
So you're proposing something like:

```python
def isReadingFor(self, *tags):
    return self.tag in tags

def isStatusFor(self, *tags):
    return self.tag in tags
```

(and eventually making it snake-case 😄 )

beauremus, 2021-05-12:
Your example will work for allowing something like `.isReadingFor(2,3,4)`. 🎉

beauremus, 2021-05-12:
If we want to accept `.isReadingFor([1,2,3])` then we need to flatten the input. I believe that could be done with

```python
import itertools
def isReadingFor(self, *tags):
    return self.tag in itertools.chain(tags)
```

rneswold, 2021-05-12:
I *don't* want to pass an array. I like the earlier example. 👍🏻

commit 030e62f2 (2021-05-18): feat: enhance `.isReadingFor()` and `.isStatusFor()`

These two methods can now take more than one tag for testing.

This commit closes #25.
closed 2021-05-18 by rneswold