fermi-ad/acsys-python #26: Generate RPM for deployment

beauremus, 2021-05-18:
Instrumentation requests that this library be installable via yum rpm.

rneswold, 2021-05-18:
Wouldn't `pip` or `poetry` install it in a more Pythonic way? If we create the `.rpm`, are we going to have to include stuff to make it recognized by the local Python installation?

beauremus, 2021-05-18:
It appears we can simply build for the rpm target. Since it uses the same process we do for building the tarball, I assume it will unpack in the same way. Maybe that's a bad assumption?

https://docs.python.org/3/distutils/builtdist.html#creating-rpm-packages

rneswold, 2021-05-18:
The `acsys` tarball contains things like an `acsys.egg-info` subdirectory. I don't know if `pip` has a registry or if it simply scans the directory in `PYTHONPATH` to see what's installed.

Would `rpm` users know where to unpack the tarball? Would they be able to `rpm`-install any dependencies?

beauremus, 2021-05-28:
I misunderstood this request. It has to do with `acnetd` and the `ACSys-Erlang` environment.

closed 2021-05-28 by beauremus