fermi-ad/acsys-python #31: Use Data Classes to more easily manage our classes

beauremus, 2021-06-01:
[Data Classes](https://docs.python.org/3.7/library/dataclasses.html) provide a simple and reliable way to implement a class with a standard set of features. Here's a demo, https://youtu.be/vBH6GRJ1REM.

This is dependent on a minimum Python version of 3.7.

rneswold, 2021-06-01:
The video also mentions an `attr` package, which may work for Python 3.6.

beauremus, 2021-06-01:
> It’s rigorously tested on Python 2.7, 3.5+, and PyPy.

So let's explore using `attrs`.

rneswold, 2021-06-01:
`ItemData` and `ItemStatus` are two classes that come to mind that might benefit from this. Maybe the event class in the SYNC library, too. I don't think `DPM` or `DPMContext` would benefit.