fermi-ad/acsys-python #36: Settings role stoped working, 1.0.0 version error

kjhazelwood, 2021-06-18:
I had been using a acsys script to make settings with the role "testing" successfully for the last few days. Today when I try the same code I get a [17 -17] console class error when running `dpm.enable_settings(role="testing")`. I tried upgrading acsys from 0.11.4 to 1.0.0. Now I get the following stacktrace trying to connect to ACNET:

```
Traceback (most recent call last):
  File "loss_generator.py", line 225, in <module>
    LossGenerator(args.config_file_path,
  File "loss_generator.py", line 74, in __init__
    devices_detuner.detune_devices(enable_settings)
  File "loss_generator.py", line 194, in detune_devices
    acsys.run_client(run)
  File "/Users/kjh/.local/lib/python3.8/site-packages/acsys/__init__.py", line 818, in run_client
    loop.run_until_complete(client_fut)
  File "/Users/kjh/opt/anaconda3/lib/python3.8/site-packages/nest_asyncio.py", line 96, in run_until_complete
    return f.result()
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/futures.py", line 178, in result
    raise self._exception
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/tasks.py", line 280, in __step
    result = coro.send(None)
  File "/Users/kjh/.local/lib/python3.8/site-packages/acsys/__init__.py", line 795, in __client_main
    await main(con, **kwargs)
  File "loss_generator.py", line 110, in run
    async with acsys.dpm.DPMContext(connection) as dpm:
  File "/Users/kjh/.local/lib/python3.8/site-packages/acsys/dpm/__init__.py", line 783, in __aenter__
    await self.dpm.connect()
  File "/Users/kjh/.local/lib/python3.8/site-packages/acsys/dpm/__init__.py", line 377, in connect
    await asyncio.wait_for(con_fut, 2000)
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/tasks.py", line 483, in wait_for
    return fut.result()
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/futures.py", line 178, in result
    raise self._exception
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/tasks.py", line 282, in __step
    result = coro.throw(exc)
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/base_events.py", line 986, in create_connection
    infos = await self._ensure_resolved(
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/base_events.py", line 1365, in _ensure_resolved
    return await loop.getaddrinfo(host, port, family=family, type=type,
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/base_events.py", line 825, in getaddrinfo
    return await self.run_in_executor(
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/futures.py", line 260, in __await__
    yield self  # This tells Task to wait for completion.
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/tasks.py", line 349, in __wakeup
    future.result()
  File "/Users/kjh/opt/anaconda3/lib/python3.8/asyncio/futures.py", line 178, in result
    raise self._exception
  File "/Users/kjh/opt/anaconda3/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/Users/kjh/opt/anaconda3/lib/python3.8/socket.py", line 918, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
OSError: Int or String expected
```

I've downgraded to 0.11.4 but the [17 -17] persists.

rneswold, 2021-06-19:
For v1.0.0, don't pass the ACNET connection to `DPMContext`. It now takes an optional port value.

beauremus, 2021-06-21:
Is there any chance you have the logger on and know what DPM you connected to? We are investigating now.

kjhazelwood, 2021-06-21:
@beauremus 
```
2021-06-21 10:22:07 INFO: connected to ACSys with handle %43712
2021-06-21 10:22:07 INFO: using DPM task: DPMD@DPM07
2021-06-21 10:22:07 INFO: DPM returned list id 61771
```

beauremus, 2021-06-21:
This turned out to be a bug in how the username is resolved to a console_user_id in the control system database. A user who was in the settings database was removed from the Fermilab system recently, and when I resynced the settings database with the DPMs, it caused an issue with resolving the console_user_id from a user name.

Removing the user from the settings database is the quick fix. The long-term fix is to handle the exception of not finding the given user.

closed 2021-06-21 by beauremus