fermi-ad/acsys-python #38: Inconsistent settings logging

kjhazelwood, 2021-06-28:
Perhaps this is more of a DPM question rather than an acsys issue, but I noticed some inconsistencies in settings logging during my study last week. While using the "testing" settings role connected to DPM01 the settings log showed the task as my machine name with no user name recorded. While using role "testing" connected to `VITR01` the task was recorded as "TCP39", user name "KJH". And when using role "reads_loss_study" connected to the DPM pool my settings were recorded as "ad1323" with no username recorded. 

What is the expected behavior? I think I'd like to see the kerberos principal name recorded in any case.

[image]

[image]

rneswold, 2021-06-29:
I think `VIRT01` is running the most recent code (we use it for testing new features.) I'll let @charlieking65  give a more detailed answer because he is the only one of us that is familiar with the DPM code and what version is running where.

rneswold, 2021-06-29:
@zongweiyuan, are there any conditions in the setting server code where the user name gets cleared? Python code can't make a setting unless the Kerberos credentials are good and @charlieking65 says DPM always sends the user name.

Also, is there a condition where the task name gets replaced by the machine name?

charlieking65, 2021-06-29:
I'm looking into it but am currently on vacation until next Tuesday.  Brian has also noticed some inconsistencies on the console side too, so it's unclear where the problem is at.

zongweiyuan, 2021-07-06:
Only java setting can provide extra information like user name, server name, desktop name. Java setting means using java class UserSettingSouce to make setting, in this class I collected all those information and packed into setting object. If Kyle's settings are made from python code, it has no such information with setting.

zongweiyuan, 2021-07-06:
I need to update SETSDB java oac to add kerberos information for python setting. acsys maybe need to make change too.

beauremus, 2021-07-06:
@charlieking65 Will be able to speak with more knowledge on how this works now, but Python sends a Kerberos credential to DPM, written in Java, that puts an entry into SETSDB on Python's behalf.

zongweiyuan, 2021-07-09:
Setting sever SETSDB receives setting in acnet message format. It not gets user Kerberos directly, it gets a int console_user_id coming from console_account table. It locates at 28 byte in acnet message after int node and before displayName.

zongweiyuan, 2021-07-09:
@rneswold task name is not replaced by machine name, for java it shows oac name or blank if not. Java got machine name from other place.

rneswold, 2021-07-15:
I know splitting this issue across projects is awkward, but please continue it in the DPM project.

closed 2021-07-15 by rneswold