fermi-ad/acsys-python #39: What is ItemData.cycle represent?

kjhazelwood, 2021-07-14:
I see in the `README.md` for `acsys.dpm` a attribute of `ItemData` called `cycle`. I dont see it available in 0.11.4, I presume its not available until v1.0? What does `cycle` represent? Iteration count of the `dpm.replies()` genenerator? Or an actual beam cycle id (that would be even better, but not sure what that would be from)?

rneswold, 2021-07-14:
It was to be the 15 Hz Booster cycle number.

In `v1.x`, we removed this field because `ItemData` now always has an array of `(stamp, value)` pairs. For slow rates, the array will have one element. Faster than 15 Hz will switch to the FTP protocol and `ItemData`'s array will have more than one element.

The cycle number doesn't easily fit in this model (since you could ask for 100Hz data which can span cycles.)

rneswold, 2021-07-14:
I built a test version in which `ItemData` gets the `.cycle` value.

For slow rates, it appears to be a millisecond timestamp from the Epoch. For fast rates, it's just a packet counter. I don't mind committing this code, but maybe we can convince @charlieking65 to make DPM return the actual cycle number of the data.

kjhazelwood, 2021-07-14:
Thanks for the explanation. For my purposes, even at FTP rates, a 15Hz cycle_id would be very beneficial. Data synchronization even at high data rates usually only has to be attributed to the same Booster injection. I think FTP itself does a pretty good job for synchronizing data on the mS level.

rneswold, 2021-07-15:
We need to get the DPM project onto GitHub so we can transfer this issue to it. 😄

beauremus, 2021-07-15:
I think it's fine to leave this here, but I created a DPM repo here if you want to transfer this issue https://github.com/fermi-controls/data-pool-manager.

rneswold, 2021-07-15:
aeca61c496739d14a585f5c1626d93d4b0e8abf3 makes the cycle value available in `ItemData`'s `.cycle` property. The rest of this issue will be handled in the issue mentioned above.

closed 2021-07-15 by rneswold