fermi-ad/acsys-python #40: Is automatic switching of datasource preferrable?

kjhazelwood, 2021-07-14:
I'm not sure the DPM switching data source from GETS32 to FTP when request is >=15Hz is always the desired behavior. Often, nodes seem to get bogged down if requesting FTP data for multiple devices which we often do (BLMs, BPMs). I think most of these nodes are capable of 60Hz GETS32 (I'm not certain). While the timestamps are preferable with FTP, I dont think you want to automatically switch to FTP unless there is a way for the user to disable this. Is there a DRF property that allows a >=15Hz request in which the user can tell the DPM to not use FTP? I know there is a DRF property to enable FTP, is there the opposite?

I think its also important that the ItemData.meta attribute specify the datasource so the user knows this.

What happens when the user requests a data rate, say 130Hz that is not a supported FTP class on a particular node? Does the DPM alter the request to match the nodes capabilities? If so, does the user get any notification of this? I think I would prefer the DPM error on not supported FTP class and make the user specify a supported rate.

I guess on the user end I don't see a benefit of the auto switching, I like having to be explicit so I know what I'm getting. But I can see how switching to FTP may help DPM, larger packets, less transactions?

rneswold, 2021-07-15:
Historically (i.e. the console world), 15Hz or slower meant RETDAT and faster than 15Hz meant FTP. (GETS32 inherited RETDAT's limits.) Now, with PIP-II on the horizon, 20Hz is the new 15Hz and becomes the deciding factor.

> What happens when the user requests a data rate, say 130Hz that is not a supported FTP class on a particular node? Does the DPM alter the request to match the nodes capabilities?

Good question. I believe DPM sends the class code query, but I'm not sure how it handles the information (whether it silently downgrades the rate or simply returns an error.) Another question for @charlieking65 tomorrow!

In the TCLK event string, there is a field to indicate the request must be handled by a hardware decoder, software decoder, or either. We could add a field for periodic events that indicate whether the rate must be honored or can be reduced.

Personally I like the common interface for the data format and that DPM seamlessly switches. We've gone decades where FTPMAN and GETS32 required different APIs to use which is why there's really only one console app that does FTP and every other app does 15Hz or slower collection.

Plus, if we come up with an improved protocol, only DPM needs to support it -- the clients will reap the benefits without any changes.

beauremus, 2021-07-15:
In my experience supporting the Python client, people want it to "just work". I deeply understand how this can create problems. I think we should definitely be transparent if the quality of service is changing so I like the idea of a field that indicates the data source.

@kjhazelwood would you be ok with DPM trying its best unless you set the `strict` flag like @rneswold mentioned and then if your request can't be fulfilled then you get an error?

In an ideal world, none of this matters if FEs are implemented in a single way and all data is available at kHz (HRM v2 everywhere!) over a single protocol. I know we won't have this any time soon but my hope is to push for this unity through ACORN.

I'm transferring this issue to the https://github.com/fermi-controls/data-pool-manager project.