fermi-ad/acsys-python #50: Allow Manual Closing of DAQ Context Manager Outside of the Context

beauremus, 2022-04-03:
We need a way to manually close an AsyncContextManager outside of the context manager. For example, in a parent process that spawns data acquisition threads.

We recommended that a user store the context in parent memory and call `.cancel()` at a later time. This doesn't work for several reasons. First, `DPMContext` returns a `DPM` object, not a Context Manager and `DPM` doesn't implement a close method. Second, even if we change the API to return the manager as well, calling `.aclose()` or `dpm._shutdown()` results in `RuntimeError: aclose(): asynchronous generator is already running`.

zongweiyuan, 2022-04-04:
I like to try. first I need to repeat this situation.

rneswold, 2022-04-04:
Thanks, Zongwei. I'm experimenting with something at the moment. Hopefully it's a solution.

PR #53 (2022-10-27): V0.x pull request (attempt 2)
Here's another attempt at #51, which was meant to fix #50.
closed 2022-10-27 by beauremus