fermi-ad/acsys-python #54: SYNC service missing clock events

kjhazelwood, 2022-06-02:
Often when using the `SYNC` service, it fails to relay clock events I've subscribed to. I've verfied the events are being seen by `D33`. This seems to happen regardless of which `SYNC` node Im connected to (`CLX68E`, `CLX56E`, `CLX38E`). I can sometimes convince myself it is working better while connected to LAN instead of Wifi. The same code has worked well in the past.

Im using acsys v0.11.4. Im monitoring device FTP readings simultaneously and combining the `DPM` and `SYNC` streams with `airstream.stream.combine.merge()`

Example code:
```
async def _monitor_beam(self, connection: acsys.Connection):
    async with acsys.dpm.DPMContext(connection, dpm_node="DPM01") as dpm:
        device_requests = {
            "qxr": (0, "I:QXRMI@P,720H,TRUE"),
            "beam": (1, "I:BEAM@P,720H,TRUE"),
            "p_gain": (2, "I_QEGAIN@E,39,E,0"),
            "i_gain": (3, "I_QGMEM@E,39,E,0"),
        }

        event_requests = {"spill_start_event": f"E,39,E,0", "spill_end_event": f"E,36,E,0"}

        logger.debug(f"Device requests count: {len(device_requests)}")
        logger.debug(f"Event requests count: {len(event_requests)}")

        await dpm.add_entries(list(device_requests.values()))
        await dpm.start()

        tclk = acsys.sync.get_events(connection, list(event_requests.values()))

        # Create a new stream from dpm and tclk generators
        all = stream.combine.merge(dpm, tclk)

        async with all.stream() as replies:
            async for reply in replies:
                try:
                    if isinstance(reply, ItemData):
                        logger.debug(f"Device {reply.meta['name']}, {reply.stamp}")

                        if isinstance(reply.data, list):
                            spill[list(device_requests.keys())[reply.tag]].append(reply)
                        else:
                            spill[list(device_requests.keys())[reply.tag]] = reply.data
                    elif isinstance(reply, ItemStatus):
                        logger.warning("%s: %s" % (device_requests[list(device_requests.keys())[reply.tag]][1], reply.status))
                    elif isinstance(reply, ClockEvent):
                        logger.debug(f"Event {reply.event:X}, {reply.stamp}")
                except Exception as e:
                        logger.warning(traceback.format_exc(e))

acsys.run_client(self._monitor_beam)
```

rneswold, 2022-06-02:
From the limited time I've looked at it, it seems to be working:

```python
#!/usr/bin/env python3                                                                                                                                           

import acsys
import acsys.sync

async def main(con):
    events = acsys.sync.get_events(con, ["E,39,E,0", "E,36,E,0"])
    async for event in events:
        print(event)

acsys.run_client(main)
```

yields

```
[clx39:~/py-acsys]$ tests/sync.py 
<TCLK event:0x39, stamp:2022-06-02 19:35:58.686000+00:00>
<TCLK event:0x36, stamp:2022-06-02 19:36:03.036000+00:00>
<TCLK event:0x39, stamp:2022-06-02 19:36:59.272000+00:00>
<TCLK event:0x36, stamp:2022-06-02 19:37:03.622000+00:00>
<TCLK event:0x39, stamp:2022-06-02 19:37:59.861000+00:00>
<TCLK event:0x36, stamp:2022-06-02 19:38:04.211000+00:00>
<TCLK event:0x39, stamp:2022-06-02 19:39:00.454000+00:00>
<TCLK event:0x36, stamp:2022-06-02 19:39:04.804000+00:00>
<TCLK event:0x39, stamp:2022-06-02 19:40:01.064000+00:00>
<TCLK event:0x36, stamp:2022-06-02 19:40:05.414000+00:00>
<TCLK event:0x39, stamp:2022-06-02 19:41:01.667000+00:00>
<TCLK event:0x36, stamp:2022-06-02 19:41:06.017000+00:00>
<TCLK event:0x39, stamp:2022-06-02 19:42:02.260000+00:00>
<TCLK event:0x36, stamp:2022-06-02 19:42:06.610000+00:00>
<TCLK event:0x39, stamp:2022-06-02 19:43:02.855000+00:00>
<TCLK event:0x36, stamp:2022-06-02 19:43:07.205000+00:00>
```

which matches with what `D33` reports.

I'm concerned about two of your device rates: 720Hz ! Could it be that the DPM generator is using two much CPU? Also, a lot of stream combiners I've seen claim unfair pulling. For instance, you have the DPM stream first so as long as there is data to pull, it won't pull from your TCLK stream. Can you switch the two so it pulls from TCLK first? Since the TCLK stream rarely generates data, it won't starve out the DPM stream.

kjhazelwood, 2022-06-02:
Thanks Rich!

I also suspect the stream combiner, I'm writing a test, I'll share the results

kjhazelwood, 2022-06-02:
Swapping the `DPM` and `SYNC` streams in the merge did make things better. However, occasionaly the `SYNC` service is disrupting the `FTP` data, so its not perfect, but better. I tried moving the `SYNC` and `DPM` streams to separate threads, writing to a common queue and that alleviated the problem but the code is not pretty. At least I know the issue now, thanks.

rneswold, 2022-06-02:
Closing this, since it's not a bug in `SYNC`. It's a shame the stream combinator library isn't more performant.

closed 2022-06-02 by rneswold