fermi-ad/acsys-python #65: pip repository is no longer public

beauremus, 2022-11-16:
This was noticed by the datalogger-to-ml project, whose GitHub workflow build depends on having access to our pip repo.

We should figure out how to accommodate this.

rneswold, 2022-11-18:
Our `--extra-url` wasn't working? Blocked by the firewall?

beauremus, 2022-11-18:
That's correct. The build from GitHub using `pip install --extra-index-url https://www-bd.fnal.gov/pip3 acsys` produces
```
ERROR: Could not find a version that satisfies the requirement acsys (from versions: none)
ERROR: No matching distribution found for acsys
```
This context may be useful
https://github.com/fermi-controls/datalogger-to-ml/issues/38
https://github.com/fermi-controls/datalogger-to-ml/issues/39

closed 2025-08-19 by beauremus