fermi-ad/acsys-python #76: acsys.status.Status: [1 -45]

gopikaops, 2024-02-06:
I am getting this error 
I am collecting acsys-py data for list of variables.
I get all the data for one variable and then - I get this error.
I have to rearrange my list - put the variable for which I already got data below and start the process again.
Then I get data for the first variable in my list and then for the 2nd this error occurs - 

I am running code on ACORN Jupyter notebook in EAF. 

```
(base) [gopikab@jupyter-gopikab gmps-ml]$python3 collection_beam_condition_paramete.py 
L:TUNRAD@E,52,E,0
Feb 6, 2024
!
[]
L:V5QSET,LLRF
L:V5QSET
LLRF
lost connection with ACSys
Traceback (most recent call last):
  File "/home/gopikab/gmps-eda/acsys-py-data/gmps-ml/collection_beam_condition_parameters.py", line 71, in <module>
    general_getter(device, ':'+node)
  File "/home/gopikab/gmps-eda/acsys-py-data/gmps-ml/collection_beam_condition_parameters.py", line 43, in general_getter
    return acsys.run_client(wrapper)
  File "/opt/conda/lib/python3.9/site-packages/acsys/__init__.py", line 824, in run_client
    return loop.run_until_complete(client_fut)
  File "/opt/conda/lib/python3.9/site-packages/nest_asyncio.py", line 99, in run_until_complete
    return f.result()
  File "/opt/conda/lib/python3.9/asyncio/futures.py", line 201, in result
    raise self._exception
  File "/opt/conda/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/opt/conda/lib/python3.9/site-packages/acsys/__init__.py", line 798, in __client_main
    con = await Connection.create()
  File "/opt/conda/lib/python3.9/site-packages/acsys/__init__.py", line 438, in create
    await con._connect(proto)
  File "/opt/conda/lib/python3.9/site-packages/acsys/__init__.py", line 419, in _connect
    res = await proto.xact(buf)
  File "/opt/conda/lib/python3.9/site-packages/acsys/__init__.py", line 289, in xact
    return _decode_ack(await ack_fut)
  File "/opt/conda/lib/python3.9/site-packages/acsys/__init__.py", line 149, in _decode_ack
    return (_ackMap.get(buf[2] * 256 + buf[3], _throw_bug))(buf)
  File "/opt/conda/lib/python3.9/site-packages/acsys/__init__.py", line 146, in _throw_bug
    def _throw_bug(_): raise status.ACNET_BUG
acsys.status.Status: [1 -45]
```

beauremus, 2024-02-06:
Please update to v0.12.6.
`pip install --no-cache-dir "acsys[settings]" --extra-index-url https://www-bd.fnal.gov/pip3`

gopikaops, 2024-02-07:
That did not work.
Requirement was already satisfied.

beauremus, 2024-02-07:
Sorry. I should have mentioned that you need to uninstall first. Can you try that?

gopikaops, 2024-02-07:
Thanks.
I did uninstall acsys and ran the command you gave.
I have a new error: [1 -34]

Also to note
I get all the data for 2 devices.
and then it fails.

This is consistent.

```
AD142074-MLT:gmps-ml gopikab$ python3 collection_gallery_ambient_parameters.py 
M:OUTTMP,FNPL
M:OUTTMP
FNPL
Feb 7, 2024
Getting Data
G:HUMID,DCE38
G:HUMID
DCE38
Feb 7, 2024
Getting Data
G:WCHILL,DCE38
G:WCHILL
DCE38
lost connection with ACSys
Traceback (most recent call last):
  File "/Users/gopikab/Documents/vimin-ml/gmps-ml/collection_gallery_ambient_parameters.py", line 71, in <module>
    general_getter(device, ':'+node)
  File "/Users/gopikab/Documents/vimin-ml/gmps-ml/collection_gallery_ambient_parameters.py", line 43, in general_getter
    return acsys.run_client(wrapper)
  File "/Users/gopikab/Library/Python/3.9/lib/python/site-packages/acsys/__init__.py", line 824, in run_client
    return loop.run_until_complete(client_fut)
  File "/Users/gopikab/Library/Python/3.9/lib/python/site-packages/nest_asyncio.py", line 90, in run_until_complete
    return f.result()
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/asyncio/futures.py", line 201, in result
    raise self._exception
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/Users/gopikab/Library/Python/3.9/lib/python/site-packages/acsys/__init__.py", line 798, in __client_main
    con = await Connection.create()
  File "/Users/gopikab/Library/Python/3.9/lib/python/site-packages/acsys/__init__.py", line 438, in create
    await con._connect(proto)
  File "/Users/gopikab/Library/Python/3.9/lib/python/site-packages/acsys/__init__.py", line 430, in _connect
    raise sts
acsys.status.Status: [1 -34]
```

closed 2024-02-07 by rneswold